MKFirmware/MarlinKimbra4due #2: Stepper

Wurstnase, 2015-04-22:
There are some bugs in the stepper on the last patch. You should update it. This is also now much faster.
https://github.com/Wurstnase/Marlin4Due/commit/1c7857b7a0b9acf2b38a22ea1238bd4a4db567b4

MagoKimbra, 2015-04-22:
Wow... thank's....
One question, but I do not understand how do you program these timers, based on what works?
As for everything else to me is quite understandable, but these things timers I just can not understand them ..

Wurstnase, 2015-04-22:
The interrupt runs at 42MHz. So if I want a stepping rate of 5000 steps/s (this is ~ 62.5mm/s with 1/16 and 80steps/mm) you need to set the counter to 42.000.000 / 5.000 = 8.400. Now the interrupt counts from 0 up to 8.400 and then it overflows and restart the stepper-isr. 

While accelerating or decelerating you need to calculate these values every step (or second on doublestepping or forth on quadstepping) to have a smooth move.

MagoKimbra, 2015-04-22:
Ok thank's for explain...

Wurstnase, 2015-05-03:
your fastio is way too slow to achieve high step rates.

MagoKimbra, 2015-05-03:
Hi Wurstnase, I have tried to do it as your own, but the card is blocked and nothing works.
I can not understand why, so instead it works fine on the Cartesian, the question remains on the delta.

Wurstnase, 2015-05-03:
I guess you set a pullup by a WRITE and not a PULLUP (SET_PULLUP).

MagoKimbra, 2015-05-04:
Ok i put your version fastio. All pin function, Bed, Hotend, Fan, Enable, but the motor not move. I read pin for sensor, endstop.
Only motor not move. The motor is block when enabled, and not block when disabled.
Where is the problem?

Wurstnase, 2015-05-05:
OK, which kind of stepper driver do you use?

MagoKimbra, 2015-05-05:
i have this board http://www.3dartists.org/ Alligator with stepper driver Texas Instrument DRV8825.

Wurstnase, 2015-05-05:
You shouldn't change that part. In the past this was STEP_ADD and STEP_IF_COUNTER. This makes normally enough delay between Step-High and Step-Low.

If that is not enough we should take other ways than a delay in an interrupt.

MagoKimbra, 2015-05-05:
I tried to do it that way, I used the part that was already there for stepper toshiba, but not enough, engines vibrated.

Wurstnase, 2015-05-05:
Anyway, it is way better to use that part of code and only put 1 delay between step_adds and step_if_counters and not a delay for every step.

closed 2015-08-10 by MagoKimbra