MKFirmware/MarlinKimbra4due #3: SPI_PIN and SPI_CHAN on Ramps4Due

designermx2, 2015-05-06:
Can't compile software when choose Ramps4Due.
With no declaration SPI_PIN :

The error show me:
HAL.cpp: In function 'void spiBegin()':
HAL.cpp:115:26: error: 'SPI_PIN' was not declared in this scope
        g_APinDescription[SPI_PIN].pPort,
                          ^
HAL.cpp: In function 'void spiInit(uint8_t)':
HAL.cpp:138:29: error: 'SPI_CHAN' was not declared in this scope
     SPI_ConfigureNPCS(SPI0, SPI_CHAN, SPI_CSR_NCPHA | SPI_CSR_CSAAT |
                             ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h:291:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/Device/ATMEL/sam3xa/include/sam3xa.h:44,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/Device/ATMEL/sam3.h:59,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/Device/ATMEL/sam.h:198,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/libsam/chip.h:25,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/cores/arduino/Arduino.h:42,
                 from HAL.h:36,
                 from HAL.cpp:31:
HAL.cpp:139:20: error: 'spiDueDividors' was not declared in this scope
       SPI_CSR_SCBR(spiDueDividors[spiClock]) | SPI_CSR_DLYBCT(1));
                    ^
HAL.cpp:139:7: note: in expansion of macro 'SPI_CSR_SCBR'
       SPI_CSR_SCBR(spiDueDividors[spiClock]) | SPI_CSR_DLYBCT(1));

If use #define SPI_PIN               49

The error will be:

HAL.cpp: In function 'void spiInit(uint8_t)':
HAL.cpp:138:29: error: 'SPI_CHAN' was not declared in this scope
     SPI_ConfigureNPCS(SPI0, SPI_CHAN, SPI_CSR_NCPHA | SPI_CSR_CSAAT |
                             ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h:291:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/Device/ATMEL/sam3xa/include/sam3xa.h:44,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/Device/ATMEL/sam3.h:59,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/Device/ATMEL/sam.h:198,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/libsam/chip.h:25,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/cores/arduino/Arduino.h:42,
                 from HAL.h:36,
                 from HAL.cpp:31:
HAL.cpp:139:20: error: 'spiDueDividors' was not declared in this scope
       SPI_CSR_SCBR(spiDueDividors[spiClock]) | SPI_CSR_DLYBCT(1));
                    ^
HAL.cpp:139:7: note: in expansion of macro 'SPI_CSR_SCBR'
       SPI_CSR_SCBR(spiDueDividors[spiClock]) | SPI_CSR_DLYBCT(1));
       ^

designermx2, 2015-05-06:
Sorry... I'm using a Smart Ramps on a Due. Its the same style of Ramps 1.4

Thanks.

giutrec, 2016-03-15:
You can read a pins.h file to examine a routine for definitions of SPI_PIN and SPI_CHAN, from line 133 onwards.

BUT you must define a pin SDSS (the alternatives are SDSS = 4 or 10 or 52 or 59 or 77) on your BOARD_XXX.h for the display and SPI_PIN, SPI_CHAN will be defined automatically.

closed 2016-05-27 by MagoKimbra