MKFirmware/MarlinKimbra4due #10: G30 A AND G29

prophetmaster, 2015-10-26:
if I put a figure greater than 55 in bed_printer_radius  nothing works properly neither has nor the G29 g30.

My delta have a 200mm diameter bed before it working with marlin...

MagoKimbra, 2015-10-27:
There is talk of the radius of the bed so the maximum positions in all directions ... I have a mirror diameter of 150 mm and I put 70 and works well, I do not understand why you can not put more than 55. What happens?

prophetmaster, 2015-10-27:
I managed to make it work, it was my proximity sensor that was creating sorry bugs...

prophetmaster, 2015-10-28:
hello I always have a problem with my Delta, it moves oddly and I do not know why?

here is my actual configuration on the machine, looking from above

[image]

Kimbra marlin in my setup is as follows :

#define Z_PROBE_OFFSET {0, 15, -6.3}

#define MANUAL_X_HOME_POS 0
#define MANUAL_Y_HOME_POS 0

#define HOTEND_OFFSET_X {-15.0, 26.0, 0.0, 0.0} // (in mm) for each hotend, offset of the hotend on the X axis
#define HOTEND_OFFSET_Y {-7.5, 0, 0.0, 0.0} // (in mm) for each hotend, offset of the hotend on the Y axis
#define HOTEND_OFFSET_Z {0.0, 0.0, 0.0, 0.0} // (in mm) for each hotend, offset of the hotend on the Z axis

What is wrong?

prophetmaster, 2015-10-30:
hello, i have checked your code in marlin it is not right for the probing ! if you define a printable radius exemple 100mm and your probe are offset of 18mm in Y axis. the printer move not 100mm+18mm it move 100mm and print 118mm probing point !!!

prophetmaster, 2015-10-30:
and in G29 you homing and not reseting bed level matrix... if you restart G29 it add offset and crash the head !

MagoKimbra, 2015-10-31:
Sorry... Ok i checked it and fix....

prophetmaster, 2015-10-31:
no problem, just 2 week for resolving the problem lol you want my modified code ?

prophetmaster, 2015-10-31:
i have added to Configuration_Delta.h this : 

// If no change in autocalibration move
#define DELTA_CALIBRATION_CHANGE_PRECISION 0.001 //mm

// Bed Printer radius
#define DELTA_PROBABLE_RADIUS 80 // mm
#define BED_PRINTER_RADIUS 100 // mm

and uncommented in conditionals.h : 

 // Radius for probe
  //#define DELTA_PROBABLE_RADIUS BED_PRINTER_RADIUS

in Marlin_main.cpp : 

i have modified adjusting number by DELTA_CALIBRATION_CHANGE_PRECISION ;-)

prophetmaster, 2015-11-14:
here is a video of the problem on Delta machine and G29 

https://www.youtube.com/watch?v=yKKvpHmU3dc

prophetmaster, 2015-11-14:
i have partialy identified the problem, it is the offset of the prob if it is not in negative...

MagoKimbra, 2015-11-14:
Hi, I don't understand..

prophetmaster, 2015-11-14:
I'll try to be more clear in my explanation, presumably the problem is present only in the G29 and G30 not in A. It looks like he does not just calculate the probe position or center of the efflector.

Have you seen the video ?

MagoKimbra, 2015-11-14:
I've seen the video, the behavior is very strange ...
You have the 4.1.2 or the dev version?
What kind of probe you?

prophetmaster, 2015-11-14:
The version is the dev but it's also about the probe 4.1.2 is a LJ8A3-2-Z / BX 2mm Tubular Inductive Proximity Sensor Switch Detector DC NPN

prophetmaster, 2015-11-14:
The probe works well I tested it on another delta without problems

prophetmaster, 2015-11-16:
hi, I can be found the source of the problem. correct me if I'm wrong: 

marlin_main.cpp in the void calibrate_print_surface function (float z_offset) you set this to 20 probing point in the array probe_z float, probe_bed_array [20]; 

I think he would rather this: 

probe_z float, probe_bed_array [(* AUTO_BED_LEVELING_GRID_POINTS AUTO_BED_LEVELING_GRID_POINTS)]; 

and modify the following line also: 

while ((probe_done == false) and (probe_count <(AUTO_BED_LEVELING_GRID_POINTS AUTO_BED_LEVELING_GRID_POINTS *))); Does it seem to you just like analysis?

closed 2016-05-27 by MagoKimbra