MKFirmware/MarlinKimbra4due #12: Ramps FD v1 Incorrect temperature

lollokara, 2015-11-25:
The temperature shows up as 4-5 times the ambient temperature, same printer with same firmware (a part from the DEFINE Board_name) but different board and microcontroller (ramps 1.4 with mega) is reading right temps.

Drakelive, 2016-03-11:
MarlinKimbra4DUE and RampsFD board work only with Arduino DUE.
RampsFD v1 has some design problems and should be modified, otherwise you risk damaging Arduino DUE.

giutrec, 2016-03-15:
Hi,
All Ramps FD V1.x  is bad designed and has serval bug it's need more fix to work correcty, otherwise you can damage your Arduino DUE.
In this link you find a guide to fix more problems
http://forums.reprap.org/read.php?219,424146

closed 2016-05-27 by MagoKimbra