MKFirmware/MarlinKimbra4due #14: SPI error when enable max6675 termocupole on Arduino Due

giutrec, 2016-03-01:
Hi,
When i try to enable a max6675 support on arduino due board, it's fail to compile and get this output

`sketch/module/temperature/temperature.cpp: In function 'void tp_init()':
sketch/module/temperature/temperature.cpp:1000:15: error: 'SS_PIN' was not declared in this scope
       pinMode(SS_PIN, OUTPUT);
               ^
sketch/module/temperature/temperature.cpp: In function 'int read_max6675()':
sketch/module/temperature/temperature.cpp:1238:5: error: 'SPCR' was not declared in this scope
     SPCR = BIT(MSTR) | BIT(SPE) | BIT(SPR0);
     ^
In file included from sketch/module/temperature/../../base.h:14:0,
                 from sketch/module/temperature/temperature.cpp:21:
sketch/module/temperature/temperature.cpp:1238:16: error: 'MSTR' was not declared in this scope
     SPCR = BIT(MSTR) | BIT(SPE) | BIT(SPR0);
                ^
sketch/module/temperature/../../module/macros.h:8:21: note: in definition of macro 'BIT'
 #define BIT(b) (1<<(b))
                     ^
sketch/module/temperature/temperature.cpp:1238:28: error: 'SPE' was not declared in this scope
     SPCR = BIT(MSTR) | BIT(SPE) | BIT(SPR0);
                            ^
sketch/module/temperature/../../module/macros.h:8:21: note: in definition of macro 'BIT'
 #define BIT(b) (1<<(b))
                     ^
sketch/module/temperature/temperature.cpp:1248:5: error: 'SPDR' was not declared in this scope
     SPDR = 0;
     ^
In file included from sketch/module/temperature/../../base.h:14:0,
                 from sketch/module/temperature/temperature.cpp:21:
sketch/module/temperature/temperature.cpp:1249:18: error: 'SPSR' was not declared in this scope
     for (; !TEST(SPSR, SPIF););
                  ^
sketch/module/temperature/../../module/macros.h:9:22: note: in definition of macro 'TEST'
 #define TEST(n,b) (((n)&BIT(b))!=0)
                      ^
sketch/module/temperature/temperature.cpp:1249:24: error: 'SPIF' was not declared in this scope
     for (; !TEST(SPSR, SPIF););
                        ^
sketch/module/temperature/../../module/macros.h:8:21: note: in definition of macro 'BIT'
 #define BIT(b) (1<<(b))
                     ^
sketch/module/temperature/temperature.cpp:1249:13: note: in expansion of macro 'TEST'
     for (; !TEST(SPSR, SPIF););
             ^
sketch/module/temperature/temperature.cpp:1255:18: error: 'SPSR' was not declared in this scope
     for (; !TEST(SPSR, SPIF););
                  ^
sketch/module/temperature/../../module/macros.h:9:22: note: in definition of macro 'TEST'
 #define TEST(n,b) (((n)&BIT(b))!=0)
                      ^
sketch/module/temperature/temperature.cpp:1255:24: error: 'SPIF' was not declared in this scope
     for (; !TEST(SPSR, SPIF););
                        ^
sketch/module/temperature/../../module/macros.h:8:21: note: in definition of macro 'BIT'
 #define BIT(b) (1<<(b))
                     ^
sketch/module/temperature/temperature.cpp:1255:13: note: in expansion of macro 'TEST'
     for (; !TEST(SPSR, SPIF););
             ^
exit status 1`

Do you can help me?

MagoKimbra, 2016-03-01:
try now!!

giutrec, 2016-03-01:
Ok, thanks it's good compile, but when is uploaded on board  system make a bootsplash screen and freeze, blank display and unexecute any serial command :(

giutrec, 2016-03-08:
Close because it's fixed!

closed 2016-03-08 by giutrec