MKFirmware/MarlinKimbra4due #17: M100_FREE_MEMORY_WATCHER desn't work

esenapaj, 2016-03-02:
if enable FAST_PWM_FAN in Configuration_Feature.h, it causes compiling error.

error message:
AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\sd\SDFat.cpp:4283:15: error: type mismatch with previous external decl of 'int* SdFatUtil::__brkval' [-fpermissive]

   extern int* __brkval;

               ^

In file included from AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\sd\../../base.h:45:0,

                 from AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\sd\SDFat.cpp:21:

AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\sd\../../module/MK_Main.h:244:16: error: previous external decl of 'void* __brkval' [-fpermissive]

   extern void *__brkval;

                ^

MagoKimbra, 2016-03-02:
Fix!!

closed 2016-03-02 by MagoKimbra

esenapaj, 2016-03-03:
I'm grateful for quick disposal, but still error occures.
error message:
AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\MK_Main.cpp.o: In function `PS_PGM':

AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\../module/communication/communication.h:28: undefined reference to `__brkval'

AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\MK_Main.cpp.o: In function `gcode_M100()':

AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module/MK_Main.cpp:5065: undefined reference to `__brkval'

collect2.exe: error: ld returned 1 exit status

MagoKimbra, 2016-03-03:
Sorry, but M100 function only 8 bit...