MKFirmware/MarlinKimbra4due #18: ADVANCE doesn't work

esenapaj, 2016-03-02:
if enable ADVANCE in Configuration_Feature.h, it causes compiling error.

error message:
In file included from AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\base.h:38:0,

                 from MarlinKimbra4due.git\trunk\MK4due\MK4due.ino:197:

AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module/sanitycheck.h:378:8: error: #error DEPENDENCY ERROR: Missing setting STEPS_MM_E

       #error DEPENDENCY ERROR: Missing setting STEPS_MM_E

        ^

if I set the STEPS_MM_E(ex. #define STEPS_MM_E 836) in Configuration_Feature.h, it still causes compiling error.

error message:
AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\motion\stepper.cpp: In function 'void TIMER0_COMPA_vect()':

AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\motion\stepper.cpp:800:5: error: 'OCR0A' was not declared in this scope

     OCR0A = old_OCR0A;

     ^

MagoKimbra, 2016-03-02:
Ok Chek!!

MagoKimbra, 2016-03-04:
I fixed!!

closed 2016-03-04 by MagoKimbra

esenapaj, 2016-03-04:
thanks for quick fix, but still error occures.
error message:
AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os  -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10608 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\n\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\system/libsam" "-IC:\Users\n\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\system/CMSIS/CMSIS/Include/" "-IC:\Users\n\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\system/CMSIS/Device/ATMEL/" "-IC:\Users\n\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\cores\arduino" "-IC:\Users\n\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\variants\arduino_due_x" "-IC:\Programs\arduino\libraries\LiquidCrystal\src" "-IC:\Users\n\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\libraries\Wire" "C:\Users\n\AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\motion\stepper.cpp" -o "C:\Users\n\AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\motion\stepper.cpp.o"
AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\motion\stepper.cpp: In function 'void TIMER0_COMPA_vect()':

AppData\Local\Temp\build41e4efdcf85178b85a086bd08f467a14.tmp\sketch\module\motion\stepper.cpp:800:5: error: 'OCR0A' was not declared in this scope

     OCR0A = old_OCR0A;

     ^