MKFirmware/MarlinKimbra4due #27: standard laser machine

mkeyno, 2016-03-13:
hi @MagoKimbra , thanks  for sharing the codes , however I coud'nt find  active low laser function in your codes , 
actually I made cnc laser with  https://github.com/TurnkeyTyranny modified marlin code  which is now updated with Hanken https://github.com/HakanBastedt/Marlin4Due/tree/Laser for due board , we discus with couple of cool guys about how the most code not consider readiness the tools in codes, and always suppose the tools is up and ready , for example in plasma machine it is necessary  ,the  torch is initiated and ready for cut , also in laser machine , laser power unit always  check the cooling system and power of the beam , and I think  this readiness should be check in loop function , it is standard way which software such as Mach3 do
I really appreciated if you could check those code as well as my due shield for cnc laser and have your comments ,

MagoKimbra, 2016-03-17:
Ok check and let you know as soon as possible ..

mkeyno, 2016-03-17:
thanks @MagoKimbra , I've check your codes  also check it with  new code of @HakanBastedt ,he combine his code with TurnkeyTyranny and got fine result , also I suggest define new definition for G0 & G1 code for laser option , because most g-code software generator , don't generate m0 and m5 for laser control 
also can you tell me which software do you use for generating g-code

daa5933, 2016-03-17:
Mkeyno, I use inkscape with the 305engineering plug-in.  It works well and is simple to use

mkeyno, 2016-03-18:
hi @daa5933 , I know 305engineering plug-in but it is not very good option for real laser machines , for example if you use the laser for cut thin metal you need dwell time to pierce the material or sometime you need check or command peripheral during the job

nextime, 2016-04-03:
On standard 8 bit MarlinKimbra i'm proposing a pull request to integrate the laser code from Turnkey Tyranny (https://github.com/MagoKimbra/MarlinKimbra/pull/113), after that, i will move to port the same patch also here for MK4due

mkeyno, 2016-04-03:
thanks @nextime Franco, I read your repository , it is really awesome to find talented people in Github ,hope  all other participant in this topic can  join the pull

mkeyno, 2016-04-03:
#https://github.com/MagoKimbra/MarlinKimbra/pull/113#issuecomment-204879376

closed 2016-04-03 by mkeyno