MKFirmware/MarlinKimbra4due #36: Arduino Due (copy) Native USB port no longer recognised Win8 x64 after flashing MK4due

bruce356, 2016-08-16:
If I load / flash other 32 bit versions of Marlin like Repetier or Wurstnase - Marlin4Due there is no problem, but with MK4due windows no longer recognises the native port, the programming port is fine. I am using the latest version of MK4due. Due combined with Radds v1.5.
Regards.

MagoKimbra, 2016-08-16:
Thank's i check this...

bruce356, 2016-08-17:
Sorry MagoKimbra, I may have been at fault (not sure), I simply flashed this "master" version without making any configuration changes.  I am very new to this 32 bit version of marlin and may I say I am very impressed, thank you for the hard work and sharing.
I subsequently configured and downloaded the latest "SHORT_BUILD_VERSION "4.2.88_dev" version from http://www.marlinkimbra.it/ and this does not cause any problems with none recognition of native port (or programming port).
thanks and regards - bruce