MKFirmware/MarlinKimbra4due #43: Extruder 2 spinning with every Z movement!

adizag, 2017-07-06:
Hi
Just updated firmeare uding the configuration tool. I use singlenozzle configuration with 2 extruders on prusa-like printer.
STRANGE PROBLEM: **Every time Z is moving the Extruder 2 is moving too!** but when moving the extruder 2, there is no movement of Z axis....

**Help will be appreciated!**

MagoKimbra, 2017-07-06:
Do you configure firmware with configurator on line or in manual mode?

adizag, 2017-07-07:
Hi

I used the online configurator.
Tried also mk4duo, but it is the same...

MagoKimbra, 2017-07-07:
So the problem with Z_TWO_STEPPER was activated, there was a bug in the configurator. Open configuration_overall.h and look for Z_TWO_STEPPER and disable it. Or by downloading it with the configurator, click on z one stepper.

adizag, 2017-07-07:
OK - *Got it solved:*
My dual Z steppers are using ONLY ONE DRIVER, so in the configurator tool,
when i choosed DUAL Z STEPPER, it's put  "Extruder 1" as default value for Z2
Stepper socket. that's why when moving Z I had additional extruder move: Z2
moved from Z driver, and the extruder moved from "extruder 1" pin allocated
for Z2... :)

THANKS ANYWAY!!!

general question: is there a problem using MK4Duo FW for a simple ramps
1.3? I don't think it's 32 bit...

Thanks again,
Adi

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
בלי
וירוסים. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_3701660066328646103_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

MagoKimbra, 2017-07-07:
MK4duo is a new version a MarlinKimbra for 8 bit and 32 bit...

adizag, 2017-07-07:
Will it work WELL on 16 bit? or shall I install MK?
(I know it is operating my printer noow, but will it be good for it?)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
בלי
וירוסים. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

____________________________________________
"יש המטביעים את צרותיהם - ויש המלמדים אותן לשחות..."
"אין מקום שהוא רחוק מדי"